zhiayang/mikoto #5: BLE has no service

ImMrMa, 2022-02-24:
Hi, thank you so much for sharing your mikoto repo, I printed the mikoto pcb and loader zmk software in the pcb. But I can't connect the hid ble sevice, seems the ble signal has no advertised service

zhiayang, 2022-02-25:
i can't tell if this is a hardware problem or a software issue (with zmk). are you able to load up a simpler firmware (eg. just something that makes a BLE beacon) and see if the antenna works?

check the component placement and soldering as well. which revision did you print?

ImMrMa, 2022-02-25:
I use the revision 5.20 for print. And flash the bootloader and zmk you mentioned. Are there any recommended programs to flash just for test the ble.

zhiayang, 2022-02-25:
you can get get bluemicro board support package for nrf52840, and use something like this https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/advertising-beacon to test it out

ImMrMa, 2022-02-26:
I recheck my bom list. found that I use 32mhz 8pf as the crystal rather than 32mhz 12.5pf you used. Is this the main reason？I try to replace a 12.5pf crystal on it.

zhiayang, 2022-02-26:
that could be the possible culprit; afaik the HF crystal is used for timing bluetooth packets, so if the capacitance is out of spec then the frequency could deviate enough that bluetooth won't work.

you can either replace the crystal, or replace the load capacitors with something that'll work for a 8pf crystal.

ImMrMa, 2022-03-05:
After replace the crystal, I finally successfully connect to the board by ble! 
Thanks for you help !!!

closed 2022-03-05 by ImMrMa