zhiayang/mikoto #6: What to choose for layer stackups and other choices for PCB manufacturing?

Ardakilic, 2022-03-19:
Hello,

I want to try this board, but I'm lost at ordering step. 

While ordering from JLCPCB, what do I have to choose for these layer stackup choices?

[image]

An example screenshot or list of choices would have been awesome. Even with providing the assets, I can't enable the SMT assembly.

Thanks in advance!

zhiayang, 2022-03-20:
L1 = F_Cu
L2 = In1_Cu
L3 = In2_Cu
L4 = B_Cu

Stackup = JLC7628

As for assembly, I don't know if they have stock of all the chips (notably the charging IC and MCU itself). Last time I checked you're also limited to green PCBs for 4-layer assembly.

Ardakilic, 2022-03-20:
Thanks for this. Finally I could pass these parameters. I just wish JLC did not cost $265 for 5 PCB's BOM, and this is even missing nRF52840 itself!

[image]

I'll definitely buy the PCBs and stencil, and source the parts in the future by myself.

Thanks again!

closed 2022-03-20 by Ardakilic

zhiayang, 2022-03-21:
heck, that is really expensive. it'll be very hard to place/solder the MCU (basically impossible) once the other parts are populated though, so getting it assembled at JLC is usually all-or-nothing.