zhiayang/mikoto #7: why short reset pin with a GPIO pin?

DoudouMee, 2022-04-09:
I dont understand why connect reset pin (P0.18/RESET) to P0.14 and P0.16. These 3 GPIO are completely shorted together. If I change the schematic and free them up, can I use they 3 pins as GPIO without short P0.18/RESET to any other pin? 
[image]

zhiayang, 2022-04-09:
[image]

there is no other way to get the reset pin out from under the chip.

DoudouMee, 2022-04-09:
> [image]
> 
> there is no other way to get the reset pin out from under the chip.

soga! So, in fact I dont need a high voltage on  (P0.18/RESET) when this Soc starting up?

zhiayang, 2022-04-09:
no, the reset pin has an internal pullup on nRF chips.

DoudouMee, 2022-04-10:
I see, thanks a lot.

closed 2022-04-10 by DoudouMee