zhiayang/mikoto #10: Device don't connect

admfgonzalez, 2023-03-05:
Hi zhiayang, first of all thank you very much for sharing your design with the community :). I sent to make some mikoto 6.3 to JLCPCB and they finally arrived, but I can't configure them, I may have done something wrong in the order process, especially with the orientation of the components. I'm not very good with electronics, could you help me identify what I did or what I'm doing wrong? Thank you very much in advance for your help. The microcontroller 
[image]
[image]

zhiayang, 2023-05-25:
Sorry the super delayed response. Unfortunately I can't really help with this, it might be a manufacturing issue

Tworck, 2023-12-12:
Hey @admfgonzalez 

Ultimately we are unsure what the problem is. The yellow light is constantly on and no connection to the microcontroller is made via usb

Did you figure out the issue to your problem. We believe that we have the same issue. 

One thing that we notices is the orientation of the Y1 Oscillator.

We also used  
https://www.farnell.com/datasheets/1562852.pdf

which states that connector 2 and connector 4 should be connected to GND. Which seems to be correct for the component we used. see image.

[image]

Kicad shows it in the same way.

[image]

So technically this should be correct. Maybe the issue is somewhere else.

zhiayang, 2023-12-13:
does the board show up over SWD? there's solder pads on the back you can use to connect to a stlink, and maybe try flashing firmware/bootloader with adafruit-nrfutil or openocd.

if it shows up over SWD, i would check the usb data lines for shorts (there's pads on the back too in case the connector pins are too small).

actually -- you guys *did* flash the bootloader, right? I don't think the usb connection will do anything (certainly it won't show up as a usb drive) until you do.