zhiayang/mikoto #11: Battery charging

djbutler, 2023-05-20:
I am trying to use Mikoto with a 3.7V 400mAh lithium ion polymer battery (https://www.adafruit.com/product/3898). I am setting the charging current to 100mA in software. But charging does not seem to work correctly: as I understand it, the battery is supposed to charge to about 4.2V, but it often gets stuck between 3V and 3.5V. It seems like perhaps the precharge phase is working, but not the fast charge phase. (See: https://www.ti.com/lit/ds/symlink/bq24075.pdf).

Is there a battery that is known to charge correctly with this circuit?

zhiayang, 2023-05-25:
I've only used no-name batteries from aliexpress, and they appear to work from my own testing. Are you setting the charge current via the driver in ZMK?

djbutler, 2023-05-28:
I'm actually running the `nice!nano` build of CircuitPython, and just setting pins P0.26 and P1.15 manually with `digitalio.DigitalInOut`:

```
en1 = digitalio.DigitalInOut(board.P0_26)
en1.direction = digitalio.Direction.OUTPUT
en1.value = False

en2 = digitalio.DigitalInOut(board.P1_15)
en2.direction = digitalio.Direction.INPUT
en2.pull = None
```

My understanding was that setting a pin as an input would produce a high-impedance state. But perhaps I'm doing this incorrectly.

Thanks for your feedback.

zhiayang, 2023-06-10:
yes, this should have worked. I'm not sure why it's not working, though. It might be some incompatibility somewhere, but I can't really help with just this much to go on.