zhiayang/mikoto #12: Wasn't sure how to reach out;

WainingForests, 2023-05-23:
Hey! I'm looking to mesh this design into something that incorporates JST SH connectors. I have a reasonably mediocre understanding of pcb design especially when it comes to signaling design. 
Headers should be easy enough.

What I don't know if I can do for sure is if I can pull out 1 or 2 more extra pins. It looks like I "might" be able to pull them out via one of the internal layers. But I realize there is alot of considerations going on with your design and I don't want to end up breaking things more than I'm helping them. 

Thoughts?

zhiayang, 2023-05-25:
Extra pins as in from the nRF chip itself? I'm not sure if there's space for that, but feel free to try. As for design considerations:

1. try not to split the ground plane on layers 2 and 3, because there isn't enough space for too many stitching vias
2. I can't remember if there's a power plane on layer 3 under the chip itself; if there is, then you definitely don't want to split it. The alternative is to get rid of the plane (if it exists) and just route power manually with thick traces.
3. As long as the modifications don't go near the antenna area, it should be fine :D

WainingForests, 2023-05-30:
Thanks by the way for the response! I've been "trying" and seem reasonably successful! I did have one question. Can you think of a way to have 3.3v output from battery that CAN'T be shut of by pin13? 

I'd like to be able to toggle LEDS while not killing a trackball / trackpad / Shift register.

zhiayang, 2023-06-10:
Probably not, if it needs to be exactly 3.3v