zhiayang/mikoto #13: resistor R12 (2M) does not appear in file gerber

hvuitsme, 2023-08-27:
I checked that the Top designator part doesn't show the 2M(R12) resistor and I can't fix it

[image]
[image]

hvuitsme, 2023-08-27:
[image]

zhiayang, 2023-08-28:
since I did not provide the BOM and pick-and-place file (BOM/CPL), I don't think I can help you here.

hvuitsme, 2023-08-29:
I got the prototype in your repository
[mikoto-top-pos.csv](https://github.com/zhiayang/mikoto/files/12459930/mikoto-top-pos.csv)

Bomb file is still the same with this pos file

zhiayang, 2023-08-29:
how did you generate this file? some of the parts (like R12) are missing rotation values. that seems to be the problem.

cscott, 2024-02-07:
You should probably use the JLC Fabrication Toolkit plugin, available from KiCad's plug-in manager, to generate the production files.