zhiayang/mikoto #14: Questions about the power path

pymo, 2023-12-10:
Hi, sorry for raising an issue for this topic, but I don't find other ways to reach out.

I am making a keyboard PCB with the WCH CH582F chip, and I find your schematic a good reference for the power path part. However, there are some design questions I hope you can help me understand:

1. You seem to replace the TP4054+LDO with the BQ24075 in later revisions, what is the consideration? My MCU is 3.3V so I want to go with the TP4054+LDO solution, but I don't know if there are any disadvantages of it.

2. How necessary are the ESD Protection Diodes on D+/D-? Did you add them because you see damages happening without it?

3. You seem to replace the Schottky barrier diode from PMEG2010AEB to PMEG4010CEJ, what is the consideration? Is it OK to use a 1N5819WS instead?

Thanks in advance!

zhiayang, 2023-12-11:
1. the nrf52840 can do HV mode which uses the internal DC-DC voltage converter, which is more efficient than an LDO so in theory saves a bit of battery. otherwise you'd have to use LDO to step 3.7v battery to 3.3v.
2. just in case. it's the most frequently "touched" part, feels weird not to have esd
3. 2010 was too small (SOD523) for my old eyes, 4010 has SOD323 which is a little bigger.  i just chose a diode with a small Vf drop.

pymo, 2023-12-11:
Thanks a lot! Those answers are very helpful!

closed 2023-12-11 by pymo